skarkkai/touch-mapper #1: Braille print output should not be raised

dkager, 2017-04-01:
STR:
- Generate a map in SVG (I used scale 1:800 to get a very localized map).
- Open the SVG in something like [TactileView](http://www.tactileview.com/).
- Send to an embosser (Index Everest-D in my case).

The result is a wallpaper-like sheet of paper that is almost completely filled with dots. My uneducated guess is that this is the bottom layer of the equivalent 3D print. This doesn't work for braille or swell-paper maps because the paper itself forms the bottom layer.

Do you have some directions on how one should properly print a map on an embosser?

skarkkai, 2017-04-01:
The SVG is generated specifically for braille and swell-paper printers, and has been successfully used by some people for that purpose. Perhaps there's something funny about the area you selected. Can you give a link to your SVG file?

dkager, 2017-04-01:
File [attached](https://github.com/skarkkai/touch-mapper/files/887894/Traverse.svg.txt).
This was successfully 3D printed last year. I would expect the SVG to be blank between streets and buildings.
It could be that the editor software I used has issues with this file. Do you know what other people were using?

skarkkai, 2017-04-01:
The SVG is white between streets and buildings, and that's expected to print as blank in a braille embosser. I think this is a problem in the software you use. I assume people use whatever software comes with their printer; I have little knowledge about that.

skarkkai, 2017-04-01:
It's also possible the content of the map is not what you expect. In case you're visually impaired, I'll mention the map has in the ballpark of 50 buildings, small and large. When that map is embossed on a A4/letter paper, I could imagine it being basically a big mess of random dots.

dkager, 2017-04-01:
Interesting! I'll try fiddling with the embossing software to see if that helps. Is the map supposed to be square? The software made it cover a full sheet of A4.

Somebody visually inspected the printed map and they said they could see some 'boxes' on it, but they were almost impossible to feel. That was with buildings excluded from the map. It looks like the dot resolution is too high.
I'll close this issue because it looks like the conversion software is at fault here, not Touch Mapper.

closed 2017-04-01 by dkager

skarkkai, 2017-04-01:
The map is always square, yes.

I'd be glad to hear how you end up solving the printing problem, and if there is a way the SVG could be improved to make the process easier for others. I could also add SVG printing instructions to Touch Mapper's help page.

dkager, 2017-04-01:
I appreciate the brainstorming we did today. Maybe this project is what will convince me to try my hand at 3D printing.

The software I mentioned (TactileView) has its own file format, but I don't think it's an open standard so SVG seems to be the best choice. I'll update this issue if I find a good way to emboss these maps.

Speaking of help, would you care for a Dutch translation?

skarkkai, 2017-04-01:
A Dutch translation would be welcome. I guess you'd be comfortable just creating another `tm.json`, based on https://github.com/skarkkai/touch-mapper/blob/master/web/locales/en/tm.json?

dkager, 2017-04-02:
Yep, I'll have a look at it.

dkager, 2017-04-03:
The cause is that the SVG contains a white 'boottom layer' rectangle:

```
<rect fill="rgb(100%, 100%, 100%)" height="299.1470896057348" width="288.424" x="-144.212" y="-154.57408960573474" />
```

Inspite of it being white, TactileView prints this shape with dots. The print looks very good if I manually remove the rectangle from the file. TactileView probably has filters to do this, but I don't know the software well enough for that.

skarkkai, 2017-04-03:
That's a good find! I think that's probably a bug in TactileView. Does it work better if you replace rgba(100%, 100%, 100%) with rgba(255, 255, 255) or #ffffff?

I could leave out the "bottom layer" rectangle entirely, and I imagine embossers would work fine without it, but visually the SVG would look kind of broken, with the background being drawn in whatever way the viewer shows transparent areas as.

dkager, 2017-04-04:
You're right, the software doesn't work correctly with RGB percentage values. Supposedly this means the whole SVG has issues, since it uses percentages everywhere. But it causes a problem only for the white bottom layer. Using `rgb(255, 255, 255)` works fine.

skarkkai, 2017-04-04:
Thanks for testing that. I'll change the color notation. The odds are TactileView isn't the only software that has trouble with the percentage notation.

skarkkai, 2017-04-17:
test.touch-mapper.org now produces SVGs with the better-working color notation. I don't mind if you'd like to test the SVGs created tehre now emboss as they should on your embosser.

dkager, 2017-04-17:
Awesome! Will give this a spin when I get back to work after the holiday.

dkager, 2017-04-21:
The new SVGs look good to me.

skarkkai, 2017-05-01:
These fixes are now in production.