skarkkai/touch-mapper #2: Connection points for multi-part maps

dkager, 2017-04-01:
Here's a random idea for the roadmap:
In 3D printed maps, allow to print connection points so that you can physically connect multiple parts of a map together. I'm thinking about some kind of male-female connection points on the edges.
Of course this requires reasonably accurate 3D prints. It could help to create very large maps, e.g. A3 or bigger.

skarkkai, 2017-04-01:
Connecting multipart maps is a real issue, and what you suggest is a possible solution. However given that the base of the maps is very thin (to minimize 3D printing time), it's not possible to have connection points outside of the map area without causing a gap between the map parts. This would only work well if either the base is made much thicker, or the map parts are like large jigsaw puzzle pieces. Either could be done eventually.

dkager, 2017-04-01:
The jigsaw idea sounds super cool. :)
You could have a square cut-out on the edge of one part and a square protrusion on the edge of the next part. But then the software has to know how many parts you are going to be making so it can finish with nice straight edges.
I guess the better workflow for that is to create a huge map and then specify in how many fixed-size 'blocks' you want to cut it.

Again, this is just a far-fetched suggestion. I don't know enough about 3D printing to know if this is even possible.

skarkkai, 2017-04-01:
Doing this is definitely possible and it's also something that is regularly suggested. But it's uncertain to what extent anyone would actually use such a feature. Given the implementation effort is pretty large, I'll postpone implementation until I have a much better idea of concrete real-life needs.