skarkkai/touch-mapper #6: Print trees

lorenzoperone, 2018-03-05:
It seems to be not possible to print trees, we would like to print urban parks and the trees position could be usefull, is it possible to implement it?

jmtaysom, 2018-04-04:
The underlying data wouldnt be able to support this. OSM doesnt capture individual trees so there wouldnt be a way to convert it into the final model.

skarkkai, 2018-04-04:
Strictly speaking OSM has a way to mark individual trees: https://wiki.openstreetmap.org/wiki/Tag:natural%3Dtree. The tag is likely rarely used, but I understand the idea of OSM is to support all reasonable use cases, including certainly tactile maps, so there's no reason this couldn't be done in a sane manner. Wouldn't be especially hard either. Not sure if many people are interested in this feature however.

lorenzoperone, 2018-04-25:
@skarkkai do you think it could be added to a next release?

skarkkai, 2018-04-27:
@lorenzoperone I'm afraid not, unless someone provides a pull request, as that's a pretty niche need.