skarkkai/touch-mapper #7: Add braille labels to maps

jmtaysom, 2018-04-04:
I just printed a map for a friend and a couple times she had to ask for the name of the street that she was touching to help her figure out the map. I dont know how helpful it would be on the maps with buildings but it should be pretty easy to find space for labels when there arent any buildings. I would be interested in helping implement this.

skarkkai, 2018-04-04:
There are many possible designs for braille labels for streets, including:

1. along the street
2. horizontally next to the street, perhaps where it crosses the map edge (if it ever does)
3. a separate printout that has a numbered list of street names, and only the numbers are rendered as braille on the map, probably on top of rectangular columns
4. like 3 but instead of a separate printout have an electronic list of numbered street names

I prefer option 4. Touch Mapper maps have persistent URLs that could be printed as QR codes on stickers and attached to the underside of the maps. That way a blind user can fairly easily see the electronic list of street names at any time by scanning the QR code with his or her phone.

An electronic list of street names could also include eg. a rough explanation about the location of the street on the map. Similarly, there could also be a list of points of interest (POI).

I understand it would often suffice to know the names and rough location (eg. "near top of the map") of the largest streets on the map, making it optional to include any new elements on the printout at all. I actually started this feature, but never completed it. It hasn't been pushed to github (for no reason).

I'll be happy to participate if you're inclined to work on this.

skarkkai, 2018-04-04:
My partner is blind and has given feedback on these options, which gives me confidence to think option 4 has merit. Further feedback on the ideas as well as printed prototypes is always helpful though.

jmtaysom, 2018-04-04:
I just got feedback and 4 was her preference too.

TTWNO, 2021-08-23:
Did anything ever happen with this? I'm curious as I had this idea to create a readable tactile map from OSM with braille lables. How close would you say you are to having anything working?

If you have the labels go up-down parrallel to the road, then you only need enough width to fit two braille dots.

Let me know.

codeofdusk, 2023-08-02:
Any updates on this? I'd like to see either option 3 or 4.

LeoLee7, 2026-04-06:
> Digital descriptions are now done, and can be seen below the 3D map preview.

Thank you so much for the update! I created a tool to manually annotate STL files with braille labels: https://leolee7.github.io/Annotate_Braille/. Please let me know if I can contribute to this amazing project. I'm currently a PhD student working on assistive agents for BLV users and I'd love to contribute or collaborate. Thanks!