skarkkai/touch-mapper #8: init.sh fails

matkoniecz, 2018-11-28:
I am planning to make some tactile maps and hopefully I can avoid starting from scratch. Thanks for sharing your work! Unfortunately, I encountered one problem

```
mateusz@grisznak:~/Desktop/tmp/touch-mapper$ ./init.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  133M  100  133M    0     0  2970k      0  0:00:45  0:00:45 --:--:-- 2956k
./init.sh: line 14: ../stl-export-blender.patch: No such file or directory
```

I am not sure what and why went wrong, so far I just followed readme and I see that script downloaded somthing. I will investigate more tomorrow.

matkoniecz, 2018-11-29:
curiously, just running it again passed this problem

matkoniecz, 2018-11-29:
Simply running it again allows it to continue.

matkoniecz, 2018-11-29:
I started to debug it:

first run:
```
mateusz@grisznak:~/Desktop/tmp/touch-mapper$ bash -x init.sh 
+ set -e
+ BLENDER_DL_URL=http://mirror.cs.umn.edu/blender.org/release/Blender2.78/blender-2.78b-linux-glibc219-x86_64.tar.bz2
+ BLENDER_VERSION=2.78
++ basename http://mirror.cs.umn.edu/blender.org/release/Blender2.78/blender-2.78b-linux-glibc219-x86_64.tar.bz2
+ blender_basename=blender-2.78b-linux-glibc219-x86_64.tar.bz2
+ [[ ! -f blender-2.78b-linux-glibc219-x86_64.tar.bz2 ]]
+ curl -O http://mirror.cs.umn.edu/blender.org/release/Blender2.78/blender-2.78b-linux-glibc219-x86_64.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  133M  100  133M    0     0  1961k      0  0:01:09  0:01:09 --:--:-- 2240k
+ tar xf blender-2.78b-linux-glibc219-x86_64.tar.bz2
+ ln -sfn blender-2.78b-linux-glibc219-x86_64 blender
+ cd blender
+ patch -p0 -s
init.sh: line 14: ../stl-export-blender.patch: No such file or directory
```

As crashing line is in the if

```
blender_basename=$(basename $BLENDER_DL_URL)
if [[ ! -f $blender_basename ]]; then
    curl -O $BLENDER_DL_URL
    tar xf $blender_basename
    ln -sfn ${blender_basename%.tar.bz2} blender
    (cd blender && patch -p0 -s <../stl-export-blender.patch)
fi
```

it means that rerunning will not crash, but that (almost certainly crucial line) is NOT executed

skarkkai, 2018-11-29:
Just remove the patch line. That patch doesn't exist anymore, so the line is obsolete.

matkoniecz, 2018-11-29:
Thanks for diagnosing! I created PR fixing this.

PR #12 (2018-11-29): remove applying nonexisting patch
fixes #8
commit 0d35bdc0 (2018-11-29): remove applying nonexisting patch

fixes #8
closed 2018-11-29 by skarkkai