skarkkai/touch-mapper #13: OSM2World compile fails

matkoniecz, 2018-11-29:
```
+ make osm2world
cd OSM2World && ant
Buildfile: /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml

help:
     [echo]   TARGETS:
     [echo]      help    => print this help
     [echo]      clean   => removes files in build/ and bin/
     [echo]      jar     => builds an optimized version of OSM2World
     [echo]      jardbg  => builds a debug build of OSM2World
     [echo]      release => builds a release version of OSM2World
     [echo]      javadoc => generates the API docs (javadoc)

main:
     [echo]   Now running default target jar

compile:
    [mkdir] Created dir: /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/bin
    [javac] Compiling 304 source files to /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/bin
    [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.7
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:10: error: cannot find symbol
    [javac] 	public Set<Point> points = new HashSet<>();
    [javac] 	           ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:12: error: cannot find symbol
    [javac] 	public Set<Point> borderCrossings = new HashSet<>(); // TODO
    [javac] 	           ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/PoiObject.java:8: error: cannot find symbol
    [javac] 	public Point center;
    [javac] 	       ^
    [javac]   symbol:   class Point
    [javac]   location: class PoiObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/console/package-info.java:4: error: cannot find symbol
    [javac] import javax.annotation.ParametersAreNonnullByDefault;
    [javac]                        ^
    [javac]   symbol:   class ParametersAreNonnullByDefault
    [javac]   location: package javax.annotation
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/console/package-info.java:1: error: cannot find symbol
    [javac] @ParametersAreNonnullByDefault
    [javac]  ^
    [javac]   symbol: class ParametersAreNonnullByDefault
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/ObjectInfoManager.java:55: error: cannot find symbol
    [javac] 			poi.center = new Point(mapNode.getPos().x, mapNode.getPos().z);
    [javac] 			                 ^
    [javac]   symbol:   class Point
    [javac]   location: class ObjectInfoManager
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:20: error: cannot find symbol
    [javac] 		points.add(new Point(segment.p1.x, segment.p1.z));
    [javac] 		               ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:21: error: cannot find symbol
    [javac] 		points.add(new Point(segment.p2.x, segment.p2.z));
    [javac] 		               ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] Note: /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/viewer/view/FileDrop.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 8 errors
    [javac] 1 warning

BUILD FAILED
/home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml:23: The following error occurred while executing this line:
/home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml:34: Compile failed; see the compiler error output for details.

Total time: 9 seconds
Makefile:5: recipe for target 'osm2world' failed
make: *** [osm2world] Error 1
```

full log: https://gist.github.com/matkoniecz/2d40ec762b448817641d01205cca3977

there is an earlier failure `google-api-python-client 1.7.4 has requirement httplib2<1dev,>=0.9.2, but you'll have httplib2 0.9.1 which is incompatible.` but I would expect it to not affect osm2world compilation

skarkkai, 2018-11-29:
Looks like java 9 needs a little fix. Try https://github.com/skarkkai/touch-mapper/tree/fix-java9-osm2world-build. That should brings in javax.annotation.ParametersAreNonnullByDefault which perhaps is the root problem.

matkoniecz, 2018-11-29:
Good news! 2 `ParametersAreNonnullByDefault` errors are gone with that patch!

But it still fails:

```
+ make osm2world
cd OSM2World && ant
Buildfile: /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml

help:
     [echo]   TARGETS:
     [echo]      help    => print this help
     [echo]      clean   => removes files in build/ and bin/
     [echo]      jar     => builds an optimized version of OSM2World
     [echo]      jardbg  => builds a debug build of OSM2World
     [echo]      release => builds a release version of OSM2World
     [echo]      javadoc => generates the API docs (javadoc)

main:
     [echo]   Now running default target jar

compile:
    [javac] Compiling 304 source files to /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/bin
    [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.7
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:10: error: cannot find symbol
    [javac] 	public Set<Point> points = new HashSet<>();
    [javac] 	           ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:12: error: cannot find symbol
    [javac] 	public Set<Point> borderCrossings = new HashSet<>(); // TODO
    [javac] 	           ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/PoiObject.java:8: error: cannot find symbol
    [javac] 	public Point center;
    [javac] 	       ^
    [javac]   symbol:   class Point
    [javac]   location: class PoiObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/ObjectInfoManager.java:55: error: cannot find symbol
    [javac] 			poi.center = new Point(mapNode.getPos().x, mapNode.getPos().z);
    [javac] 			                 ^
    [javac]   symbol:   class Point
    [javac]   location: class ObjectInfoManager
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:20: error: cannot find symbol
    [javac] 		points.add(new Point(segment.p1.x, segment.p1.z));
    [javac] 		               ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/core/map_data/object_info/WayObject.java:21: error: cannot find symbol
    [javac] 		points.add(new Point(segment.p2.x, segment.p2.z));
    [javac] 		               ^
    [javac]   symbol:   class Point
    [javac]   location: class WayObject
    [javac] Note: /home/mateusz/Desktop/tmp/touch-mapper/OSM2World/src/org/osm2world/viewer/view/FileDrop.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 6 errors
    [javac] 1 warning

BUILD FAILED
/home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml:23: The following error occurred while executing this line:
/home/mateusz/Desktop/tmp/touch-mapper/OSM2World/build.xml:34: Compile failed; see the compiler error output for details.

Total time: 7 seconds
Makefile:5: recipe for target 'osm2world' failed
make: *** [osm2world] Error 1
mateusz@grisznak:~/Desktop/tmp/touch-mapper$ g s
## fix-java9-osm2world-build...origin/fix-java9-osm2world-build
```

note that I have java8

```
mateusz@grisznak:~/Desktop/tmp/touch-mapper$ /usr/bin/java -version
java version "1.8.0_191"
Java(TM) SE Runtime Environment (build 1.8.0_191-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.191-b12, mixed mode)
```

should I update to java9 as the next step?

skarkkai, 2018-11-30:
Java 9 isn't needed. I'll take a look at this during the weekend, if not today. If you want to look into this before that, Touch Mapper probably branched from OSM2World from Dec 21, 2015 commit, or possibly the one after that. There could be some file missing from git or something like that.

Touch Mapper commit eefbd4c38a4f90619037cc0ae480e8eaa8fd2b91 "Diff from original OSM2World to git history start" should apply cleanly on the correct OSM2World commit.

matkoniecz, 2018-11-30:
```
BUILD SUCCESSFUL
Total time: 13 seconds
Finished
```

closed 2018-11-30 by matkoniecz