skarkkai/touch-mapper #15: aws credentials necessary to run local dev server

matkoniecz, 2018-11-30:
```
mateusz@grisznak:~/Desktop/tmp/touch-mapper/web$ make watch
make build
make[1]: Entering directory '/home/mateusz/Desktop/tmp/touch-mapper/web'
./pre2src.py
rm -rf build
node build.js
./create-env-js.sh dev >build/scripts/environment.js
Unable to locate credentials. You can configure credentials by running "aws configure".
./create-env-js.sh: no CloudFront distribution for 'dev-mateusz' found
Makefile:3: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/mateusz/Desktop/tmp/touch-mapper/web'
Makefile:15: recipe for target 'watch' failed
make: *** [watch] Error 2
```

I will try either reconfiguring it to make it not necessary and submit a PR or add section to readme that giving blank configuration to `aws configure` is fine and add it to a README.

matkoniecz, 2018-11-30:
Nope, blank answers to `aws configure` results in

```
mateusz@grisznak:~/Desktop/tmp/touch-mapper/web$ make watch
make build
make[1]: Entering directory '/home/mateusz/Desktop/tmp/touch-mapper/web'
./pre2src.py
rm -rf build
node build.js
./create-env-js.sh dev >build/scripts/environment.js
Unable to locate credentials. You can configure credentials by running "aws configure".
./create-env-js.sh: no CloudFront distribution for 'dev-mateusz' found
Makefile:3: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/mateusz/Desktop/tmp/touch-mapper/web'
Makefile:15: recipe for target 'watch' failed
make: *** [watch] Error 2
```

skarkkai, 2018-11-30:
What are your high level goals in setting up Touch Mapper? The server side script takes in a simple request as JSON, reads the relevant data from OSM and outputs the map model. The web & AWS portion is just a UI for creating the request JSON, and not strictly necessary to create maps.

matkoniecz, 2018-11-30:
> What are your high level goals in setting up Touch Mapper?

For start I want to change rules for generating maps (support for `area:highway`), generate several maps and 3D print them.

matkoniecz, 2018-11-30:
All done locally, I do not plan to host anything (that is why I hoped that empty credentials will be enough).

For now I am just following readme.

skarkkai, 2018-11-30:
What you have setup now is probably enough to generate maps. I'll write instructions tomorrow for how to do that.