skarkkai/touch-mapper #18: Gray scale support

Mte90, 2020-10-25:
I have a 3D laser engraver that support grayscale colors, so I can do the various levels with a single image.
It is possible to add that support?

skarkkai, 2020-10-25:
The SVG images intended for embossers have a different color for each different object type. I imagine it's possible to use those files?

Mte90, 2020-10-25:
I looked them but they are just different colors, for me it will be better on various gray scale, so it won't request manually changes.

skarkkai, 2020-10-25:
My guess is that the engraver software will treat colors as gray shades. If not, it should be pretty easy to change the colors to appropriate gray scales in any reasonable image editing program.

Mte90, 2020-10-25:
Yes with an image editor is possible, I was wondering if the tool can generate it in that way, the borders are already black so maybe is required just for the buildings and other parts.