skarkkai/touch-mapper #21: On website, add option to make walkways near-equal height to surrounding ground

jaredcoh, 2023-04-19:
I understand from the perspective of a visible disabled person, having the walkways be clearly defined is helpful. 
I do think that there is benefit in having the website provide an option to make the walkways not "go underground". I believe there is a simple in the code, where a parameter is turned from true to false, but it is not available on the website as an option to have.

skarkkai, 2023-04-23:
What do you mean by walkways going underground? What is the problem you are having with the way walkways are handled currently?

jaredcoh, 2023-04-23:
The picture below may show it here: 
[image]
This is a picture of Carnegie Mellon University, and the large X in the middle is a walkway that crosses between several buildings. On the stl that is automatically generated on the website, it appears that this is significantly lowered compared to the ground surrounding it. I was curious if there was a way to remove the significant height difference between the walkway and the ground.

Let me know if this makes sense! Thank you :)

skarkkai, 2023-04-23:
This is very likely because the entire grass field has been defined as a recreational area, which is treated by Touch Mapper the same as pedestrian footways, so they are raised. The paths might be treated as something too small to include on the map, so their area is treated like other areas that contain "nothing". So they end up lower than the grass field.

[image]

jaredcoh, 2023-04-23:
Thank you sthat makes total sense. I appreciate the explanation!

closed 2023-04-23 by jaredcoh