skarkkai/touch-mapper #22: Request for University Use

dwoods1024, 2023-10-04:
Hello Skarkkai,

My name is Dylan and I am the Liaison for the Alt Media Center for the Disabled Students' Program at UC Berkeley. We work with college students with disabilities and have found your touch-mapper software to be of great use in printing 3D Maps for blind individuals. I was told through my work colleges that there was a chance that your software may not be supported in the near future but we wanted to continue using touch-mapper to create maps for our work. Is there a possibility of working with you to support the software or perhaps if we could use the software on a private server on campus? Please let us know your thoughts.

Best,
Dylan

skarkkai, 2023-10-04:
I have no plans whatsoever to stop providing Touch Mapper. The cost of running it is very low, so I expect it to be around for many decades more.

closed 2023-10-04 by skarkkai

dwoods1024, 2023-10-05:
Hi Skarkkai,

Thank you for letting me know. Do you know if it is possible to use the repository on a private server for our University department?

Best,
Dylan

skarkkai, 2023-10-05:
It's possible, but in practice not easy, there is a fair bit of setting up to do. To my knowledge nobody has done that.

dwoods1024, 2023-10-05:
I see. I am trying to set up a repository in my GitHub Account using the Touch-Mapper I downloaded from your site. I hope it is ok but I can take it down otherwise with respect to your work. If I could use it, I promise not to sell it or use it in a way that violates any copyright law or anything else. Additionally, if you are willing, I would like to know more about how to set up Touch Mapper in a desktop/non website setting