skarkkai/touch-mapper #26: Error generating a map on the website (No such file or directory)

addic, 2024-01-11:
I can't seem to get a map generated. It goes to 60% progress and then gets stuck. In the console I see a lot of HEAD requests. The last request seems to contain an error:

> Can't convert map data to STL: [Errno 2] No such file or directory: '/home/ubuntu/touch-mapper/prod/runtime/1/map.stl'

Full request response:
> HTTP/1.1 200 OK
x-amz-id-2: WDmQBYJ9Vno6/waKH/dUoVruw7JkYL+BDYlMmYs84yuSU223v24+H40CNkGCVMc1Al+jTrcVCS0=
x-amz-request-id: QYB7MM8YKZ8DWB3R
Date: Thu, 11 Jan 2024 18:25:11 GMT
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, HEAD
Access-Control-Expose-Headers: x-amz-meta-processing-stage, x-amz-meta-error-msg, x-amz-meta-building_count
Vary: Origin, Access-Control-Request-Headers, Access-Control-Request-Method
Last-Modified: Thu, 11 Jan 2024 18:25:11 GMT
x-amz-expiration: expiry-date="Wed, 10 Jul 2024 00:00:00 GMT", rule-id="Delete after 180 days"
ETag: "d41d8cd98f00b204e9800998ecf8427e"
x-amz-storage-class: REDUCED_REDUNDANCY
x-amz-server-side-encryption: AES256
Cache-Control: max-age=8640000
x-amz-meta-error-msg: Can't convert map data to STL: [Errno 2] No such file or directory: '/home/ubuntu/touch-mapper/prod/runtime/1/map.stl'
x-amz-version-id: null
Accept-Ranges: bytes
Content-Type: binary/octet-stream
Server: AmazonS3
Content-Length: 0

skarkkai, 2024-02-12:
Fixed now. Sorry it took so long.

closed 2024-02-12 by skarkkai

GNUGradyn, 2024-09-06:
Still having the same issue, Gets stuck at 60%. Same exact thing. Here is a dump of the request
```bash
curl 'https://s3-eu-west-1.amazonaws.com/prod.maps.touch-mapper/map/data/B0af84596abf3c66/Tokyo%20Station%20Marunouchu%20Building-m100X-p100Y.stl' \
  -X 'HEAD' \
  -H 'Accept: */*' \
  -H 'Accept-Language: en-US,en;q=0.9' \
  -H 'Cache-Control: no-cache' \
  -H 'Connection: keep-alive' \
  -H 'DNT: 1' \
  -H 'Origin: https://touch-mapper.org' \
  -H 'Pragma: no-cache' \
  -H 'Referer: https://touch-mapper.org/' \
  -H 'Sec-Fetch-Dest: empty' \
  -H 'Sec-Fetch-Mode: cors' \
  -H 'Sec-Fetch-Site: cross-site' \
  -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36' \
  -H 'sec-ch-ua: "Chromium";v="127", "Not)A;Brand";v="99"' \
  -H 'sec-ch-ua-mobile: ?0' \
  -H 'sec-ch-ua-platform: "Windows"'
  ```