skarkkai/touch-mapper #27: Location cone is on STL

coinnodenl, 2024-03-03:
The location you enter will have a cone, which is on the STL, this should not be there right?

mgrundkoetter, 2024-07-11:
Would be nice to have this removable (maybe as advanced option).

Dickmaulruessler, 2024-07-11:
> Would be nice to have this removable (maybe as advanced option).

Yes, that would be great!

sirnightowl, 2024-09-10:
Agreed, this would be absolutely perfect without the cone. I think it saids "Show advanced options (also hides location marker)" so maybe it's a bug that it doesn't?

dancingbear1965, 2025-01-16:
Is there any movement of this issue? Whilst the rendered image does not show the cone, the STL file does. So it seems the 'hide location marker' aspect is not working.

daviddoria, 2025-02-24:
I also ran into this problem. It would be great to be able to remove the cone from the STL.

HJADR, 2025-02-24:
+1 for this issue.

Antwelm, 2025-04-17:
The cone is also much higher than any other feature on the map. Definition of sore thumb. Nice for those that need it, but pretty annoying for everyone else.

skarkkai, 2025-04-26:
I hear you people, I will remove this at some point. Thanks for letting me know this matters.