skarkkai/touch-mapper #32: issue with water body that seems ok in OpenStreetMap

ShortyCM, 2024-10-06:
I'm not sure what might be going on here, but I was trying to make a map of this location.  The water body looks fine in OpenStreetMap, as it appears to match the satellite image of this location on Google Maps, but for some reason the STL file output is missing a large portion of the water body.  I wonder if it might be because some of it is tagged as an intermittent water body.

43.75583646662724, -65.98087997889401

skarkkai, 2024-10-06:
Could you show screenshots of the preview (including all the UI settings), and the matching resulting STL?

ShortyCM, 2024-10-06:
Sure.  Here's the map preview, STL preview, a slicer screenshot, and the STL.  The water in the cove is completely missing, despite only a portion of it being considered intermittent water.  At least, it seems from the map view that only a portion of it is.
[water_test.zip](https://github.com/user-attachments/files/17270725/water_test.zip)